GOFIRST-Robotics/Lunabotics #15: Test Nvidia's nvblox SLAM solution

Isopod00, 2023-07-29:
Documentation: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-realsense.md

Isopod00, 2023-07-29:
How does it compare to the built-in SLAM in the ZED ROS Wrapper?

jmblixt3, 2023-09-13:
Preliminary testing has completed on zed-implement. Findings show as thought negative depth is not considered in cost map and current nvblox parameters do not contain enough sensitivity to small changes in heightmap. More exploration is needed, but I custom layer/implementation to nvblox will be needed

jmblixt3, 2023-10-02:
This node could be modified to accommodate our task
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/tree/main/nvblox_nav2

Erix4, 2023-10-03:
Here is nav2's costmap 2d code: https://github.com/ros-planning/navigation2/tree/main/nav2_costmap_2d

jmblixt3, 2023-10-13:
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/100

umnrobotics, 2023-11-03:
This has been completed in https://github.com/GOFIRST-Robotics/Lunabotics-2024/pull/87