GOFIRST-Robotics/Lunabotics #16: Create a separate app implementing av1 encoding and decoding

jmblixt3, 2023-07-29:


Isopod00, 2023-08-12:
This probably isn't super important, but it would be cool to look into / it would be a cool side project type of thing

jmblixt3, 2023-10-08:
Gatreamer apt repository for Ubuntu 22.04 uses an old build of gstreamer which doesn't have much support for AV1. It will either need to be built from source or run in a container

Isopod00, 2023-10-11:
We decided to run it in a container for now, so we should probably make sure that this camera frontend is a terminal cli app that can be run ion the container as well

Isopod00, 2024-03-24:
This is being developed in the [gstreamer-client ](https://github.com/GOFIRST-Robotics/Lunabotics-2024/tree/gstreamer-client) branch. @jmblixt3, what work is left to complete on this?

jmblixt3, 2024-03-24:
It currently is using fake cameras and just toggling back in forth based on time

jmblixt3, 2024-03-24:
We'll need to determine how we want to message pass between client and server to toggle cameras

jmblixt3, 2024-04-06:
Decision was to use ros node to do this

Isopod00, 2024-05-05:
This issue also includes making a launch file for the team laptop that starts the joystick node and also Jonathan's gstreamer camera client. This launch file is done and has been tested: it works!

jmblixt3, 2024-05-05:
Ask Peter M he was testing it

closed 2024-05-06 by Isopod00