GOFIRST-Robotics/Lunabotics #40: Define what our app needs and a preliminary design for our streaming application

jmblixt3, 2023-09-05:


Isopod00, 2023-10-01:
webapp vs. local program and what should the layout look like?

MasonJohnHawver42, 2023-10-02:
Ive written some code that got it to stream to a website on local host and it was laggy and if you reloaded the page it would some how start from the begining of the stream? So I've decided just a terminal cli app should do.

Isopod00, 2023-10-02:
@MasonJohnHawver42 As long as we can have multiple labeled buttons to switch between the different camera views (front, back, and internal/conveyor view) and the layout looks somewhat clean, then I think it's okay to choose whatever path is most efficient and easiest to implement. In terms of the lag you were experiencing, could it be because of the AV1 encoding, or were you not encoding the GStreamer feed at all? When testing a GStreamer pipeline with AV1 encoding you'll have to make sure that the hardware you're testing on actually supports AV1 encoding/decoding. Our Jetson AGX Orin does, and RTX 3000 series (or newer) Nvidia desktop/laptop GPUs do as well