GOFIRST-Robotics/Lunabotics #60: Fix setup tools in devcontainer

Isopod00, 2023-09-12:
add this 
```PYTHONWARNINGS="ignore:setup.py install is deprecated::setuptools.command.install"; export PYTHONWARNINGS```

Isopod00, 2023-09-12:
@jmblixt3 are you sure that ignoring this warning is the solution? Previously I thought the solution was to downgrade setup_tools.py to the "proper" version.

Isopod00, 2023-09-12:
actually now that I look at it I think both of these are just workarounds, the real solution is probably to just make the changes it wants: https://answers.ros.org/question/386341/ros2-userwarning-usage-of-dash-separated-install-scripts-will-not-be-supported-in-future-versions-please-use-the-underscore-name-install_scripts/#:~:text=According%20to%20this%20link%3A%20https%3A%2F%2Fanswers.ros.org%2Fquestion%2F3964...%20The%20reason%20is,install%20setuptools%3D%3D58.2.0%20and%20it%20shows%20no%20warnings%20anymore.

jmblixt3, 2023-09-12:
You can fix one of the issues like thos but others are unfixable due to ros
build system

jmblixt3, 2023-09-12:
https://github.com/ament/ament_cmake/issues/382#issuecomment-1528083515

closed 2023-09-14 by jmblixt3