GOFIRST-Robotics/Lunabotics #169: Implement / Update autonomy code in the main_control_node

Isopod00, 2024-03-01:


jmblixt3, 2024-03-17:
Add it too it's own launch file then I clude that launch file

Isopod00, 2024-03-17:
> Add it to its own launch file then I include that launch file

@jmblixt3 What do you mean by this? As far as I am aware, the easiest way to implement our autonomy logic/instructions is going to be in the main_control_node so that we can cancel it with a button press if something goes wrong. Alternately we could try using a Nav2 behavior tree instead, but that would probably take more research & time to get working. I'm not quite sure what you mean by putting this logic in its own launch file.

jmblixt3, 2024-03-17:
Nav2 behavior tree will almost certainly be the easiest way to implement and adjust this behavior