GOFIRST-Robotics/Lunabotics #177: Update To Jazzy Jalisco

jmblixt3, 2024-03-21:


Isopod00, 2024-03-21:
> This can probably happen after the competition year

Is this even possible on Ubuntu 22.04 (JetPack 6.0)?

jmblixt3, 2024-03-21:
So cool think about jetpack 6.0 is this is no longer true

jmblixt3, 2024-03-21:
> <#177 (comment)
> I see that there are more Linux distro options available now, but the only
> version of Ubuntu they list is Ubuntu 22.04? I think we are fine with
> Ubuntu 22.04 and ROS Humble though, we need to use Humble anyway as far as
> I am aware because that is what all the Nvidia ISAAC ROS packages use.

So regarding 24.04 "JetPack 6 brings the flexibility to run any Linux
Kernel." Regardless you are correct on stating the dependency on Nvidia