GOFIRST-Robotics/Lunabotics #187: Find optimal heights for the 2 static map slices

BergerKing333, 2024-03-29:
Combining the static map slices seems to be working, however, we need to mess with different heights using different recordings to see how that affects the final slice. Need another/ a better recording first though

Isopod00, 2024-04-11:
@BergerKing333 What's the status of this ticket? Is this included in your current nvblox PR, or no?

BergerKing333, 2024-04-11:
> [@BergerKing333](https://github.com/BergerKing333) What's the status of this ticket? Is this included in your current nvblox PR, or no?

No, this is more along the lines of testing the realsense dirt collection; all that needs to happen is editing 2 numbers in a config but kinda need the robot to test with to find the correct numbers to set to.

BergerKing333, 2024-04-18:
Also, mess with configs for the above ground denoising and the inflation layer configuration. need good/official recording before messing with these

jmblixt3, 2024-04-27:
This will need some adjustment for current recordings due to adjustments in the base_link transforms

closed 2024-05-14 by Isopod00