GOFIRST-Robotics/Lunabotics #295: Add code for simulating our digger and dumper in Gazebo after the new 2024-2025 CAD is exported for Gazebo

Isopod00, 2024-10-12:


Isopod00, 2025-01-21:
The drivetrain already has working simulation code but the digger and dumper do not yet

andrewtomfohrde, 2025-02-13:
I added Gazebo simulation support to the digger node by introducing a GAZEBO_SIMULATION parameter and a Gazebo publisher (Float64) for controlling the digger arm in simulation. All existing functionality remains unchanged, I only added new code to digger_node without deleting any previous code

Isopod00, 2025-02-13:
> I added Gazebo simulation support to the digger node by introducing a GAZEBO_SIMULATION parameter and a Gazebo publisher (Float64) for controlling the digger arm in simulation. All existing functionality remains unchanged, I only added new code to digger_node without deleting any previous code

I see the new branch you created yesterday but I don't see any commits on it yet, did you remember to commit & push your code at the end of yesterday's meeting?

andrewtomfohrde, 2025-02-13:
Just did, it's not done yet, I just left that comment to know where I left off