GOFIRST-Robotics/Lunabotics #311: Configure Isaac_Ros_Apriltag to use multiple cameras

BergerKing333, 2024-11-05:
Being able to use multiple cameras to detect apriltags would make early match localization quicker/easier and would allow fusing of apriltags throughout the entire match when neither of the 2 zeds are facing towards the base. We should be able to use one of our plain webcams to do this since apriltags don't require depth cameras

BergerKing333, 2024-11-05:
Im pretty sure nvidia has support for this with isaac ros 3.1 (they added multi cam support for nvblox), but if not we could frankenstein it and stitch multiple camera feeds into one with OpenCV or something then feed that to nvblox. Determining the pixel that the tag is found in we could replicate apriltag detection but this is not ideal

jmblixt3, 2024-11-05:
You could also run multiple instances of the april tag node and then combine.

Isopod00, 2024-11-08:
Jonathan's comment is the way we would want this to be done. Run one instance of isaac_ros_apriltag per camera we want searching for apriltags, and since each of these instances return TFs from the camera frame to the tag frame, it should just straight up work honestly with the Apriltag math we are already doing since I wrote it to rely on the ROS TF tree. Although, some minor modifications might be needed.

jmblixt3, 2024-11-08:
I also know there was discussion of an EKF, and they could be combined there as well potentially

Isopod00, 2025-02-07:
This is on hold until #277 is completed

closed 2026-10-12 by Michael-Foley