GOFIRST-Robotics/Lunabotics #315: Use Zed IMU to determine if robot is tipping

BergerKing333, 2024-11-08:
As discussed at the PDR, we could use the IMU's on the Zeds to determine if we are lowering the digger too low into the ground that the robot starts tilting by reading the position from the zeds and comparing them. By the time the zeds notice that we are lifted off the ground, it might be too late, but this is still a good safety feature if we have time.

Isopod00, 2025-02-14:
On Hold until #341 is completed

Isopod00, 2025-03-02:
I thought of a cool/elegant way to implement this: In the motor control node, we will want to subscribe to the "/zed2i/zed_node/pose" topic, and specifically we are interested in the pitch angle being published to this topic, as it can tell us whether we are tipping or not. Then, we basically want to modify the current Potentiometer_callback method in the motor control node to use this pitch angle and add yet another P controller (PID but just P) that uses the pitch angle as the error, to slow the digging linear actuators as the pitch angle increases. I'm not totally sure whether we want to use the direct pitch angle as the error and assume 0 degrees is always the desired state or measure the current pitch when we start digging and assume that is the desired state (subtract this from the current pitch every loop to get the error for the pitch P controller) but one of those two should work. We will want to add the output of this new P controller to the output we are setting when duty cycle has a negative goal or when position has a goal below where it is currently at (any time the digger is lowering basically, and we don't need to worry about any of this pitch stuff if it is raising).

closed 2025-04-10 by Isopod00